PX4/NuttX #1: Compilation error when CONFIG_NSOCKET_DESCRIPTORS >0

matanhavi, 2017-01-02:
Two compilation error:
1) Line 960 fs/procfs/fs_procfsproc.c
 linesize   = snprintf(procfile->line, STATUS_LINELEN, "\n%3-s %-2s %-3s %s\n", "SD", "RF", "TYP", "FLAGS");
unknown conversion type character
probably need to be changed to 
 linesize   = snprintf(procfile->line, STATUS_LINELEN, "\n%-3s %-2s %-3s %s\n", "SD", "RF", "TYP", "FLAGS");

2) Line 983 fs/procfs/fs_procfsproc.c
          linesize   = snprintf(procfile->line, STATUS_LINELEN, "%3d %2d %3d %02x",
                                i + CONFIG_NFILE_DESCRIPTORS,
                                (long)socket->s_crefs, socket->s_type, socket->s_flags)
probably need to be changed to 
          linesize   = snprintf(procfile->line, STATUS_LINELEN, "%3d %2d %3d %02x",
                                i + CONFIG_NFILE_DESCRIPTORS,
                                socket->s_crefs, socket->s_type, socket->s_flags)

matanhavi, 2017-01-05:
its seems to be fixed on 6a6a089af4da46d7b5dc7b41a727c28fde88ed3e. (April 2016)
Why its not on our branch?

davids5, 2017-01-05:
@matanhavi 

What configurations were you building? Please list the line command line here.

matanhavi, 2017-01-06:
Its custom configuration that has CONFIG_NSOCKET_DESCRIPTORS defined.
The errors are 'simple', 
I am new around and not sure how to share the fixes

davids5, 2017-01-06:
I would suspect you do not have a valid configuration. If this is a standalone Nuttx build and not part of PX4 then you should request help on the NuttX forum.

closed 2017-01-06 by davids5

matanhavi, 2017-01-07:
I think my configuration is valid.
the errors are simple
` snprintf(procfile->line, STATUS_LINELEN, "\n%3-s %-2s %-3s %s\n", "SD", "RF", "TYP", "FLAGS");`
just can't be compiled, the snprintf format is invalid, form ` %3-s`  it should be `%-3s`.
On the second error the compiler doesn't allow to format the long `(long)socket->s_crefs` as `%2d`
I am not sure how nuttx missed that, or it might be different compilation rules?
How should I fix that? at nuttx repository at bitbucket? at PX4 clone?

matanhavi, 2017-01-07:
The issue has fixed by Gregory Nutt:
[Fix on bitbucket](https://bitbucket.org/nuttx/nuttx/issues/24/compilation-error-when)
How issues like that merged into PX4 fork? periodically?

davids5, 2017-01-07:
It will be picked up in the next uptake from upstream NuttX. In the meantime you can do a PR for change that adds it to the end of nuttx-patches/wip_inflight.....patch file.