PX4/NuttX #29: fs_fat32.c fat_seek maybe return a error that shoud ok

elmagnificogi, 2018-08-18:
Use case:
open a file with readonly,then lseek to the end with offset 0
there will return a error( ENOSPC 28 "No space left on device")
so i can't get the file size with this.

TestFile:13,107,200 Byte

Code:
```
static off_t fat_seek(FAR struct file filep, off_t offset, int whence)
---------1164
if ((ff->ff_oflags & O_WROK) != 0)
{
/ Extend the cluster chain (fat_extendchain
* will follow the existing chain or add new
* clusters as needed.
*/

          cluster = fat_extendchain(fs, cluster);
        }
      else
        {
          /* Otherwise we can only follow the existing chain */

          cluster = fat_getcluster(fs, cluster);
        }
```
---------1178

when i debug i found that fat_getcluster(fs, cluster) will retrun the "0x0fffffff" that means FAT cluster end.
but in fat_seek ,there is no end deal,so there will be a error.

The break way is less than clustersize,in the loop， position Just coincidentally equal to clustersize,so it continue find next cluster chan.

---------1142
```
if (position < clustersize)
{
break;
}
```
---------1145

Solution:

add FAT end deal or make break condition wiht "position <= clustersize"

dagar, 2019-01-02:
Is this still an issue?

elmagnificogi, 2019-01-03:
> Is this still an issue?

no one reply,i have no idea