PX4/NuttX #51: fs_fat32.c fat_seek jump wrong

elmagnificogi, 2019-06-05:
at fs_fat32.c fat_seek 1239 line:
```
 /* Then get the current sector from the cluster and the offset
  * into the cluster from the position
  */

  (void)fat_currentsector(fs, ff, filep->f_pos);
```

in fact fat_currentsector will change the currentsector to the sector we seek

```
int fat_currentsector(struct fat_mountpt_s *fs, struct fat_file_s *ff,
                      off_t position)
{
  int sectoroffset;

  if (position <= ff->ff_size)
    {
      /* sectoroffset is the sector number offset into the current cluster */

      sectoroffset = SEC_NSECTORS(fs, position) & CLUS_NDXMASK(fs);

      /* The current cluster is the first sector of the cluster plus
       * the sector offset
       */

      ff->ff_currentsector = fat_cluster2sector(fs, ff->ff_currentcluster)
                           + sectoroffset;

      /* The remainder is the number of sectors left in the cluster to be
       * read/written
       */

      ff->ff_sectorsincluster = fs->fs_fatsecperclus - sectoroffset;

      finfo("position=%d currentsector=%d sectorsincluster=%d\n",
            position, ff->ff_currentsector, ff->ff_sectorsincluster);

      return OK;
    }

  /* The position does not lie within the file */

  return -ENOSPC;
}
```

you could see the code, only when position <= ff->ff_size ,the sector will be changed right.
if position > filesize,it will return err,but dont  deal with it.

if i want to write a file(size 1000) at 1500, first you seek to 1500,the seek func won change the current_sector,the current_sector will restain the last time.Then you use write,just write to the last sector not the sector 1500 bytes should be.This make the random seek and write not work and make a mess.

The solve ：
```
      int sectoroffset;

	  /* sectoroffset is the sector number offset into the current cluster */

	  sectoroffset = SEC_NSECTORS(fs, filep->f_pos) & CLUS_NDXMASK(fs);

	  /* The current cluster is the first sector of the cluster plus
	   * the sector offset
	   */

	  ff->ff_currentsector = fat_cluster2sector(fs, ff->ff_currentcluster)
						   + sectoroffset;

	  /* The remainder is the number of sectors left in the cluster to be
	   * read/written
	   */

	  ff->ff_sectorsincluster = fs->fs_fatsecperclus - sectoroffset;
```

use the code replace the func fat_currentsector in seek func.

in other situation  the fat_currentsector  maybe work right

elmagnificogi, 2019-06-05:
PS：
in write it will change the fat_currentsector , but the cache sector is not right.

only in seek func,it will reload the cache sector 

```
      /* Load the sector corresponding to the position */

      if ((position & SEC_NDXMASK(fs)) != 0)
        {
          ret = fat_ffcacheread(fs, ff, ff->ff_currentsector);
          if (ret < 0)
            {
              goto errout_with_semaphore;
            }
        }
```

julianoes, 2019-06-05:
@elmagnificogi can I suggest that you check if this has been fixed upstream:
https://bitbucket.org/nuttx/nuttx/src/master/
which is the same as the master branch in this repository: https://github.com/PX4/NuttX/tree/master

elmagnificogi, 2019-06-05:
@julianoes  i checked,no fix