PX4/NuttX #64: statfs() might lead to halt

BazookaJoe1900, 2019-10-02:
I added mavlink message, that checks the SD card status periodically (in contrast to current implementation _STORAGE_INFORMATION_, that need to be asked by the ground).
during checking my message, I tested what will happen if I remove the SD card during operation. doing that caused the mavlink data to stop.

first, do you consider that as a problem? how the system behave if SD is removed, or get malfunction during flight.
I only checked that above scenario, that I added new mavlink message. but i guess that other methods that required the SD are much more critical, for example reading mission on commander....

I think that it related to the fact that reading the status of fat32 is done by _fat_statfs(),_ that waits for semaphore _fat_semtake(fs);_

Testing code can be found at:
https://github.com/BazookaJoe1900/Firmware/tree/testing-sd_removal

(is that the right place to point to that issue?)

BazookaJoe1900, 2019-10-04:
closed in favor to https://github.com/PX4/Firmware/issues/13087

closed 2019-10-04 by BazookaJoe1900

julianoes, 2019-10-31:
Fixed by #68