PX4/NuttX #84: NuttX build fail with CLICOLOR_FORCE=1 on Mac

supergra, 2020-02-20:
On MacOS, with environment variable `CLICOLOR_FORCE=1` set in `.bashrc`, the NuttX compilation fails at several places where it parses `ls` to find lists of kconfig files. Terminal color codes get spliced into the master kconfig list, which then fails to parse correctly.

For my color configuration, it failed only for files that were incorrectly marked executable. Although these permissions appear to have been fixed in the upstream NuttX, other users might run into this issue on all files, depending on their color configuration.

My immediate solution was to move CLICOLOR_FORCE to `.bash_profile` where it belongs (or disable it altogether). However, it may take users a very long time to discover this problem. 

Another solution:
At the top of `platforms/nuttx/NuttX/apps/tools/mkkconfig.sh`, add `unset CLICOLOR_FORCE`.

I expect this change may be too specific to be accepted, but at least this will document the issue for future users.

Here's the error message:
```
.../Firmware/build/px4_fmu-v5_<model>/NuttX/apps/examples/Kconfig:57: '.../Firmware/build/px4_fmu-v5_<model>/NuttX/apps/examples/lis3dsh_reader/Kconfig' not found (in 'source ".../Firmware/build/px4_fmu-v5_<model>/NuttX/apps/examples/lis3dsh_reader/Kconfig"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
```

dagar, 2021-02-23:
We can back port the upstream fix if it applies cleanly. Otherwise we'll sync up with upstream NuttX eventually. 

Alternatively we could force these environment variables within PX4 when it builds NuttX.

Do you want to open a PR?

closed 2021-02-23 by dagar

supergra, 2021-02-23:
I'm not able to do a PR, and I'm ok with leaving it unchanged. The main goal was to get this documented somewhere to maybe save someone the couple hours of debugging it took me!