PX4/NuttX #90: "depends on IMXRT_DTCM > 0" do problems for savedefconfig

Igor-Misic, 2020-03-07:
This line does some problems while using `make px4_fmu-v4 qconfig` while saving defconfig. 
 
https://github.com/PX4/NuttX/blob/d509b1edd736cb7db509a9c86e7ffcde10c4fd45/arch/arm/src/imxrt/Kconfig#L967

```
LN: platform/board to /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4/NuttX/apps/platform/dummy
arch/arm/src/imxrt/Kconfig:968: syntax error
arch/arm/src/imxrt/Kconfig:967: invalid option
tools/Makefile.unix:595: recipe for target 'savedefconfig' failed
```

tianguozun, 2020-08-28:
when make px4_fmu-v5_default  menuconfig , it is failed, The error message is as follows：
arch/arm/src/imxrt/Kconfig:968: syntax error
arch/arm/src/imxrt/Kconfig:967: invalid option

How can I solve it?

davids5, 2020-08-28:
Please change '>' to `!=`

davids5, 2020-09-14:
@tianguozun @Igor-Misic - I can not reproduce this error. Have you tried update-ding your kconfig-conf?

Igor-Misic, 2020-09-16:
Witch kconfig-conf is used here? 
This one:
https://github.com/PX4/Firmware/blob/master/platforms/nuttx/NuttX/tools/kconfig-conf

Or kconfig-conf used for Linux itself?

davids5, 2020-09-16:
Have a look at https://bitbucket.org/nuttx/tools/src/master/

Igor-Misic, 2020-09-16:
Ok, I've run this again:

```
git clone https://bitbucket.org/nuttx/tools.git
cd tools/kconfig-frontends
sudo apt install gperf
./configure --enable-mconf --disable-nconf --disable-gconf --enable-qconf --prefix=/usr
make
sudo make install
```

/usr/bin/kconfig-conf
Is from today.

`make px4_fmu-v4 qconfig

Output:
```
[1/2] Running NuttX make qconfig for nsh
Create .version
LN: include/arch/board to /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/../nuttx-config//include
LN: include/arch/chip to arch/arm/include/stm32
LN: arch/arm/src/board to /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/../nuttx-config//src
LN: arch/arm/src/chip to arch/arm/src/stm32
LN: /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/drivers/platform to /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/../nuttx-config/drivers
LN: platform/board to /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/apps/platform/dummy
[2/2] Running make qconfig then savedefconfig for nsh
Create .version
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/../nuttx-config//include
LN: include/arch/chip to arch/arm/include/stm32
LN: arch/arm/src/board to /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/../nuttx-config//src
LN: arch/arm/src/chip to arch/arm/src/stm32
LN: /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/drivers/platform to /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/../nuttx-config/drivers
LN: platform/board to /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/apps/platform/dummy
arch/arm/src/imxrt/Kconfig:968: syntax error
arch/arm/src/imxrt/Kconfig:967: invalid option
tools/Makefile.unix:595: recipe for target 'savedefconfig' failed
make[1]: *** [savedefconfig] Error 1
FAILED: cd /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx && make --no-print-directory --silent -C /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx CONFIG_ARCH_BOARD_CUSTOM=y savedefconfig && /usr/bin/cmake -E copy /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/defconfig /home/igor/GIT/PX4/Firmware/boards/px4/fmu-v4/nuttx-config/nsh/defconfig && /usr/bin/cmake -E remove -f /home/igor/GIT/PX4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/.config
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_fmu-v4' failed
make: *** [px4_fmu-v4] Error 1
```

davids5, 2021-03-05:
kconfig-confdies not give a version so here is what I am running 

-rwxr-xr-x 1 root root 59896 Jul  9  2018 /usr/local/bin/kconfig-conf and it works fine.

Igor-Misic, 2021-03-05:
So it seems my kconfig-conf is older, maybe that is the problem. 

-rwxr-xr-x 1 root root 55856 lis   5  2017 /usr/local/bin/kconfig-conf

Igor-Misic, 2021-03-05:
Yep, just deleted /usr/local/bin/kconfig-conf and now is working with newer  /usr/bin/kconfig-conf

Thanks @davids5 !

closed 2021-03-05 by Igor-Misic