PX4/NuttX #198: Nuttx configuration for px4

HadesGuy, 2022-10-21:
Hi, 
I am following tutorial  https://www.youtube.com/watch?v=2Bvku3dnrr8 . 
On the 13:22 minute, It can be seen the commands needed for installation and configuration of Nuttx for a board - which are " ./tools/configure.sh -c nucleo-l476rg/nsh"

I would like to use the board for px4 CUAV V5+ that has processor:  STM32F765   -------------/https://docs.px4.io/main/en/flight_controller/cuav_v5_plus.html       --------------

I have changed the board location to: stm32f769i-disco/nsh  ,but this gives out the errors below:

tools/configure.bat -c stm32f769i-disco/nsh
mingw32-gcc.exe -Wall -Wstrict-prototypes -Wshadow -g -pipe -I. -DCONFIG_WINDOWS_NATIVE=y -o configure.exe configure.c cfgparser.c
ERROR: stat of (null) failed: Invalid argument
configure.exe failed

I also tried with:
 tools/configure.sh -c stm32f769i-disco/nsh

and get the following error: 

tools/configure.sh: line 142: [: /home/User: binary operator expected
tools/configure.sh: line 161: [: /home/User: binary operator expected
tools/configure.sh: line 181: [: /home/User: binary operator expected
tools/configure.sh: line 186: [: /home/User: binary operator expected
grep: /home/User: No such file or directory
grep: UserSurname/nuttx/tools/../boards/*/*/stm32f769i-disco/configs/nsh/defconfig: No such file or directory
grep: /home/User: No such file or directory
grep: UserSurname/nuttx/tools/../boards/*/*/stm32f769i-disco/configs/nsh/defconfig: No such file or directory
  Copy files
ln: target UserSirname/nuttx/tools/../Make.defs': No such file or directory
Failed to symlink /home/UserSurname/nuttx/tools/../boards/*/*/stm32f769i-disco/configs/nsh/Make.defs

I can`t figure out what I'm doing wrong. 
Any advice is appreciated. 

Thanks

davids5, 2022-10-21:
Use the https://github.com/PX4/PX4-Autopilot repo and build the px4_fmu_v5 (see https://github.com/PX4/PX4-Autopilot/blob/dff946c39a01b883fbd7967c8b9255ca02be826b/boards/px4/fmu-v5/src/manifest.c#L163)

HadesGuy, 2022-10-24:
Hi 
I have followed the https://docs.px4.io/main/en/dev_setup/building_px4.html and went trough the Ubuntu and Nuttx examples but it does not goes quite as expected. 
The Nuttx install goes like this:

cd PX4-Autopilot
make px4_fmu-v5_default   

-- PX4 version: v1.13.0-beta1-1046-gc267cf71c3
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.9.10", minimum required is "3")
-- PX4 config file: /home/sim_dim/PX4-Autopilot/boards/px4/fmu-v5/default.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m7
-- ROMFSROOT px4fmu_common
-- IO px4_io-v2_default
-- SERIAL_GPS1 /dev/ttyS0
-- SERIAL_TEL1 /dev/ttyS1
-- SERIAL_TEL2 /dev/ttyS2
-- SERIAL_TEL4 /dev/ttyS3
-- UAVCAN_INTERFACES 2
-- UAVCAN_TIMER_OVERRIDE 6
-- PX4 config: px4_fmu-v5_default
-- PX4 platform: nuttx
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "<compiler-path>" == *"cygdrive"* ]]; then cygpath -w "<compiler-path>"; else echo <compiler-path>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "<dir1>" == *"cygdrive"* ]]; then cygpath -w "<dir1>"; else echo <dir1>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "[<dir2>" == *"cygdrive"* ]]; then cygpath -w "[<dir2>"; else echo [<dir2>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "[<dir3>" == *"cygdrive"* ]]; then cygpath -w "[<dir3>"; else echo [<dir3>; fi '
/bin/sh: -c: line 0: unexpected EOF while looking for matching `''
/bin/sh: -c: line 1: syntax error: unexpected end of file
/bin/sh: -c: line 0: unexpected EOF while looking for matching `''
/bin/sh: -c: line 1: syntax error: unexpected end of file
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "<compiler-path>" == *"cygdrive"* ]]; then cygpath -w "<compiler-path>"; else echo <compiler-path>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "<dir1>" == *"cygdrive"* ]]; then cygpath -w "<dir1>"; else echo <dir1>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "[<dir2>" == *"cygdrive"* ]]; then cygpath -w "[<dir2>"; else echo [<dir2>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "[<dir3>" == *"cygdrive"* ]]; then cygpath -w "[<dir3>"; else echo [<dir3>; fi '
/bin/sh: -c: line 0: unexpected EOF while looking for matching `''
/bin/sh: -c: line 1: syntax error: unexpected end of file
/bin/sh: -c: line 0: unexpected EOF while looking for matching `''
/bin/sh: -c: line 1: syntax error: unexpected end of file
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "<compiler-path>" == *"cygdrive"* ]]; then cygpath -w "<compiler-path>"; else echo <compiler-path>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "<dir1>" == *"cygdrive"* ]]; then cygpath -w "<dir1>"; else echo <dir1>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "[<dir2>" == *"cygdrive"* ]]; then cygpath -w "[<dir2>"; else echo [<dir2>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "[<dir3>" == *"cygdrive"* ]]; then cygpath -w "[<dir3>"; else echo [<dir3>; fi '
/bin/sh: -c: line 0: unexpected EOF while looking for matching `''
/bin/sh: -c: line 1: syntax error: unexpected end of file
/bin/sh: -c: line 0: unexpected EOF while looking for matching `''
/bin/sh: -c: line 1: syntax error: unexpected end of file
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "<compiler-path>" == *"cygdrive"* ]]; then cygpath -w "<compiler-path>"; else echo <compiler-path>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "<dir1>" == *"cygdrive"* ]]; then cygpath -w "<dir1>"; else echo <dir1>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "[<dir2>" == *"cygdrive"* ]]; then cygpath -w "[<dir2>"; else echo [<dir2>; fi '
/bin/sh: -c: line 0: syntax error near unexpected token `;'
/bin/sh: -c: line 0: `if [[ "[<dir3>" == *"cygdrive"* ]]; then cygpath -w "[<dir3>"; else echo [<dir3>; fi '
/bin/sh: -c: line 0: unexpected EOF while looking for matching `''
/bin/sh: -c: line 1: syntax error: unexpected end of file
/bin/sh: -c: line 0: unexpected EOF while looking for matching `''
[... 54 lines cut ...]
  The CMAKE_ASM_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Enabling double FP precision hardware instructions
-- Configuring incomplete, errors occurred!
See also "/home/sim_dim/PX4-Autopilot/build/px4_fmu-v5_default/CMakeFiles/CMakeOutput.log".
See also "/home/sim_dim/PX4-Autopilot/build/px4_fmu-v5_default/CMakeFiles/CMakeError.log".
Error: /home/sim_dim/PX4-Autopilot/build/px4_fmu-v5_default is not a directory
make: *** [Makefile:227: px4_fmu-v5_default] Error 1

And when I look at the part:  

 Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

I am struggling  to figure out how to do this bit above - how to find the CMake file and change the path

davids5, 2022-10-24:
@HadesGuy Please see https://docs.px4.io/main/en/dev_setup/getting_started.html and use the scripts to install the tools.  I would recommend running on Ubuntu natively or use Virtual Box

HadesGuy, 2022-10-26:
Hi, I am using the scripts and using Ubuntu 18.04 on Virtual Box. 
starting from the beginning I have installed the tools show in https://docs.px4.io/main/en/dev_setup/building_px4.html 
Now i get the error: 

sim@sim-VirtualBox:~/PX4-Autopilot$ make px4_fmu-v5_default
-- PX4 version: v1.13.0-beta1-1062-g740d2fccb1
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- PX4 config file: /home/sim/PX4-Autopilot/boards/px4/fmu-v5/default.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m7
-- ROMFSROOT px4fmu_common
-- IO px4_io-v2_default
-- SERIAL_GPS1 /dev/ttyS0
-- SERIAL_TEL1 /dev/ttyS1
-- SERIAL_TEL2 /dev/ttyS2
-- SERIAL_TEL4 /dev/ttyS3
-- UAVCAN_INTERFACES 2
-- UAVCAN_TIMER_OVERRIDE 6
-- PX4 config: px4_fmu-v5_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 6.3.1
-- The C compiler identification is GNU 6.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Enabling double FP precision hardware instructions
-- drivers/px4io: ROMFS including px4_io-v2_default
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- Using C++11
-- Release build type: MinSizeRel
CMake Warning at src/modules/microdds_client/CMakeLists.txt:35 (message):
  skipping microdds_client, Micro-XRCE-DDS-Client needs to be fixed to work
  with CMAKE_VERSION 3.10.2

-- ROMFS: ROMFS/px4fmu_common
-- ROMFS:  Adding platforms/nuttx/init/stm32f7/rc.board_arch_defaults -> /etc/init.d/rc.board_arch_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_defaults -> /etc/init.d/rc.board_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_sensors -> /etc/init.d/rc.board_sensors
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_fmu-v5_bootloader.bin -> /etc/extras/px4_fmu-v5_bootloader.bin
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_io-v2_default.bin -> /etc/extras/px4_io-v2_default.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sim/PX4-Autopilot/build/px4_fmu-v5_default
[0/1175] git submodule src/drivers/gps/devices
[8/1175] git submodule platforms/nuttx/NuttX/nuttx
[12/1175] git submodule src/drivers/uavcan/libuavcan
[13/1175] git submodule src/modules/mavlink/mavlink
[14/1175] Generating Mavlink standard: ...k/message_definitions/v1.0/standard.xml
Traceback (most recent call last):
  File "/home/sim/PX4-Autopilot/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/sim/PX4-Autopilot/src/modules/mavlink/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ModuleNotFoundError: No module named 'future'
FAILED: mavlink/standard/standard.h 
cd /home/sim/PX4-Autopilot/build/px4_fmu-v5_default/src/modules/mavlink && /usr/bin/python3 /home/sim/PX4-Autopilot/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py --lang C --wire-protocol 2.0 --output /home/sim/PX4-Autopilot/build/px4_fmu-v5_default/mavlink /home/sim/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
[20/1175] Generating parameters.xml
ninja: build stopped: subcommand failed.
Makefile:227: recipe for target 'px4_fmu-v5_default' failed
make: *** [px4_fmu-v5_default] Error 1
sim@sim-VirtualBox:~/PX4-Autopilot$ make px4_fmu-v5_default

I believe I am following the tutorial correctly, but maybe I am missing something ..

HadesGuy, 2022-10-26:
The link you have posted does not contain scripts, but links. I have followed the link for toolchain.

davids5, 2022-10-26:
This one? https://github.com/PX4/PX4-Autopilot/blob/main/Tools/setup/ubuntu.sh

HadesGuy, 2022-10-28:
Hi Davis,
I have just checked - these are installed in the ubuntu.sh
Outcome is the same :(

davids5, 2022-10-28:
@dagar @julianoes Any advice here?

HadesGuy, 2022-10-31:
Now I ran: make px4_fmu-v5_default - again and got multiple errors one of which is repeating: 

                from ../../src/drivers/uavcan/libuavcan/libuavcan/src/marshal/uc_bit_array_copy.cpp:5:
/usr/include/newlib/math.h:18:3: error: '__ULong' does not name a type
   __ULong i[2];
   ^~~~~~~