PX4/NuttX #215: Hardfault: Bus fault on vector table read (stm32f103ret6) on a dronecan node

dirksavage88, 2023-01-17:
This is a dronecan application firmware that is hard faulting after the bootloader updates firmware (there is nothing occupying flash space after the bootloader). The near exact board config worked with an stm32f103zet6 prototype (only differences is that there is no fsmc, and only 1 memory region on the ret6) and I was able to get the dronecan node operational and a nuttx console going.

The board config can be found here: https://github.com/PX4/PX4-Autopilot/compare/main...dirksavage88:Firmware:aa_cannode_v1

Dronecan bootloader works as intended and I see it come up in init mode, it allows me to update the firmware via the dronecan gui tool. I also verify node status and node info when I plug in the node flashed with the bootloader. With the hardfault, it doesn't recover and the bootloader doesn't jump into firmware enough for me to get a system console (it just restarts and the node comes back up in init).

See screen snips. I am reading the hardfault in the bus monitor ascii section. 

@ryanjAA

@davids5 I believe this is similar to oliminexino stm32 you brought into Nuttx, it's just a higher density version (512kb flash, 64kb sram).

Other known issues: the nuttx hardware pinmap for the stm32f103r is missing uart4 & uart5. I've made these changes in my local nuttx submodule but it is not fixed upstream. The node has a sensor on uart4.

[image]
[image]
[image]

davids5, 2023-01-19:
Some references

The Olimex Can node was running a STM32F103RBT6 
[image]

The  STM32F103RET6 and the STM32F103ZET6 differ in pin and flash size

[image]

davids5, 2023-01-20:
I would bet the board does not have and LSE (Low speed crystal) or it is not starting

closed 2023-01-20 by dirksavage88