PX4/NuttX #306: CANHUBK3 eth0 down, not able to ping

dirksavage88, 2024-04-20:
The default state of eth0 is down, When issuing the command `ifup eth0` the interface reports 'okay' but does not init. I am unable to ping out from the device. 

[image]

PetervdPerk-NXP, 2024-04-21:
Netman manages the eth0 interface and turns it off automatically through link detection.

What network config does netman show?

dirksavage88, 2024-04-21:
Hmm, so the ping works with dhcp when the connected to my router via T1 converter, but not able to ping the NavQ plus with a static IP. 

By default I don't show a net.cfg in "/fs/microsd/". Even if I create one and echo the key val pairs followed by 'netman update -i eth0' I cant seem to get the static IP to work

DaShen96, 2024-06-21:
Hi, can you show me the "defconfig" file?

dirksavage88, 2024-06-21:
> Hi, can you show me the "defconfig" file?

https://github.com/dirksavage88/Firmware/blob/master/boards/nxp/mr-canhubk3/nuttx-config/nsh/defconfig